rambunction4330/RobotCode2022 #17: Code formatting

ADawesomeguy, 2022-01-15:
I was thinking we could use something like `astyle` or `clang-format` to automagically format code with consistent indentation according to common standards.

ADawesomeguy, 2022-01-15:
Additionally we could add CodeQL or MSVC analysis as a workflow.

ADawesomeguy, 2022-01-15:
It may also be a good idea to tag at certain points in order to provide easy points to fall back to as well milestones.

gcjurgiel, 2022-01-16:
Formalizing the formatting standards we use is a good idea, but I think the "automagic" formatting of `clang-format` is unnecessary.

ADawesomeguy, 2022-01-16:
In that case I think we should leave a code quality workflow in the repo to check code quality to adhere to standards without automatic formatting.

gcjurgiel, 2022-01-16:
That sounds good, though you'll have to show me how it works on Tuesday.

ADawesomeguy, 2022-01-17:
No problem. If you go to the repo main page there's a little green ✔️ next to the commits, signaling a successful build. For the code quality checks it'll be a similar thing except instead of checking successful build it'll check for code quality.

ADawesomeguy, 2022-01-20:
Using `clang-format` with the LLVM style flag on `HolonomicDrive.cpp`:

Original file:
```cpp
#include "rmb/drive/HolonomicDrive.h"

#include <algorithm>

namespace rmb {
  void HolonomicDrive::driveCartesian(double ySpeed, double xSpeed, double zRotation) {
    double y = std::clamp(ySpeed, -1.0, 1.0);
    double x = std::clamp(xSpeed, -1.0, 1.0);
    double t = std::clamp(zRotation, -1.0, 1.0);

    frc::ChassisSpeeds speeds = {y * getMaxVel(), 
                                 x * getMaxVel(), 
                                 t * getMaxRotVel()};
    driveChassisSpeeds(speeds);
  }

  void HolonomicDrive::drivePolar(double magnitude, double angle, double zRotation) {
    double m = std::clamp(magnitude, -1.0, 1.0);
    double t = std::clamp(zRotation, -1.0, 1.0);

    double radAngle = units::convert<units::degree, units::radian>(angle);
    driveCartesian(m * sin(radAngle), m * cos(radAngle), t);
  }
}
```

Modified file:
```cpp
#include "rmb/drive/HolonomicDrive.h"

#include <algorithm>

namespace rmb {
void HolonomicDrive::driveCartesian(double ySpeed, double xSpeed,
                                    double zRotation) {
  double y = std::clamp(ySpeed, -1.0, 1.0);
  double x = std::clamp(xSpeed, -1.0, 1.0);
  double t = std::clamp(zRotation, -1.0, 1.0);

  frc::ChassisSpeeds speeds = {y * getMaxVel(), x * getMaxVel(),
                               t * getMaxRotVel()};
  driveChassisSpeeds(speeds);
}

void HolonomicDrive::drivePolar(double magnitude, double angle,
                                double zRotation) {
  double m = std::clamp(magnitude, -1.0, 1.0);
  double t = std::clamp(zRotation, -1.0, 1.0);

  double radAngle = units::convert<units::degree, units::radian>(angle);
  driveCartesian(m * sin(radAngle), m * cos(radAngle), t);
}
} // namespace rmb
```

Diff:
```diff
--- HolonomicDrive.cpp	2022-01-20 10:06:39.000000000 -0600
+++ test.cpp	2022-01-20 10:09:09.000000000 -0600
@@ -3,22 +3,23 @@
 #include <algorithm>
 
 namespace rmb {
-  void HolonomicDrive::driveCartesian(double ySpeed, double xSpeed, double zRotation) {
-    double y = std::clamp(ySpeed, -1.0, 1.0);
-    double x = std::clamp(xSpeed, -1.0, 1.0);
-    double t = std::clamp(zRotation, -1.0, 1.0);
+void HolonomicDrive::driveCartesian(double ySpeed, double xSpeed,
+                                    double zRotation) {
+  double y = std::clamp(ySpeed, -1.0, 1.0);
+  double x = std::clamp(xSpeed, -1.0, 1.0);
+  double t = std::clamp(zRotation, -1.0, 1.0);
 
-    frc::ChassisSpeeds speeds = {y * getMaxVel(), 
-                                 x * getMaxVel(), 
-                                 t * getMaxRotVel()};
-    driveChassisSpeeds(speeds);
-  }
+  frc::ChassisSpeeds speeds = {y * getMaxVel(), x * getMaxVel(),
+                               t * getMaxRotVel()};
+  driveChassisSpeeds(speeds);
+}
 
-  void HolonomicDrive::drivePolar(double magnitude, double angle, double zRotation) {
-    double m = std::clamp(magnitude, -1.0, 1.0);
-    double t = std::clamp(zRotation, -1.0, 1.0);
+void HolonomicDrive::drivePolar(double magnitude, double angle,
+                                double zRotation) {
+  double m = std::clamp(magnitude, -1.0, 1.0);
+  double t = std::clamp(zRotation, -1.0, 1.0);
 
-    double radAngle = units::convert<units::degree, units::radian>(angle);
-    driveCartesian(m * sin(radAngle), m * cos(radAngle), t);
-  }
+  double radAngle = units::convert<units::degree, units::radian>(angle);
+  driveCartesian(m * sin(radAngle), m * cos(radAngle), t);
 }
+} // namespace rmb
```

ADawesomeguy, 2022-01-22:
Alright so #42 kinda did this so I think we could add a workflow to check if the format in certain directories adheres to the LLVM standard. To make sure it does people can run `clang-format --style=LLVM -i [file(s)]`. Closing for now.

ADawesomeguy, 2022-01-22:
I'm reopening this; @gcjurgiel mind taking a look at this workflow? We should be able to let it check our code for formatting.

https://github.com/marketplace/actions/clang-format-check

theVerySharpFlat, 2022-02-02:
this one is an oldy but a goody

ADawesomeguy, 2022-02-02:
true
wait i can close this

ADawesomeguy, 2022-02-02:
I implemented the check

closed 2022-02-02 by ADawesomeguy