rambunction4330/RobotCode2022 #18: Motorcontroller implementation structure: Templates and Inheritance

theVerySharpFlat, 2022-01-16:
Hey @gcjurgiel,

You said last meeting that you would remove templates from the motorcontroller base class. Based on your last commit, it seems like we are keeping this. Could you please share with me how you fixed the units issue + the excessively long function declarations?

Also, should the SparkMaxVelocitycontroller also inherit from the frc::MotorSaftey class like the TalonSRX does?

Thanks

gcjurgiel, 2022-01-16:
You're right, I did say we would get rid of the templates but I still liked having them, so I was using the talon classes to experiment with how to make it less gross. You can't get rid of the length in the return type of the function, but the function parameters are inside the namespace of the class, so you don't need to use the long version.

The Talons additionally don't handle conversions well so I added a unit for the native distance and velocity units the encoders return. To help this I also changed wheelDiameter to a conversion factor with a unit of DistanceUnit/radian. That way if the distance is radians, the conversion factor is just the gear ratio. If the DistanceUnit is meters, they conversion can account for both wheel diameter and gear ratio. This leverage the units library since the multiplications (or divisions depending on which direction you're converting to) should just work out to the proper units.

I'm not sure if I want continue having the motor controllers inheriting from frc::MotorSaftey. Based on what I've been looking at, I think it's unnecessary. Similarly, Im not sure if we should keep the functions I copied from frc::MotorController (frc::SpeedController) to the VelocityController class. Inverting seems useful, but the others are less so.

theVerySharpFlat, 2022-01-17:
Ok, that makes sense. I will try to implement the sparkmaxes based on the talonsrxes.

Could you please explain this code?
https://github.com/rambunction4330/RobotCode2022/blob/7c8d7fa338a5ffff9e62b663a55127b85ac33822/lib/include/rmb/motorcontrol/talon/TalonSRXVelocityController.h#L21-L24

Understand that you are creating a ratio called RawVelocity_t which is DistanceUnit/time but I am not quite sure what this line does:
https://github.com/rambunction4330/RobotCode2022/blob/7c8d7fa338a5ffff9e62b663a55127b85ac33822/lib/include/rmb/motorcontrol/talon/TalonSRXVelocityController.h#L21

I looked at the units:: library docs and I couldn't find a constructor that took 3 arguments to create a unit tag. Could you please explain this to me?

Thanks,
Aiden

gcjurgiel, 2022-01-17:
The Talon MagEncoder has 4096 encoder ticks per rotation, and returns velocity in encoder ticks per 100ms (or 1 decisecond). RawUnit declares a new angle unit that has a ration conversion of 2pi radians per ever 4096 encoder ticks (2pi/4096). The first argument is the ratio, the next is the base unit, and the third is the exponent of pi. There is another translation argument (like how you have to add 273 when covering from Kelvin to Celsius) but it is given a default value of 0 (similarly, the pi exponent also has a default of 0). You can see the definition here:

https://github.com/nholthaus/units/blob/master/include/units.h#L834-L864

A simplified (but better) explanation is [here](https://github.com/nholthaus/units#defining-new-units) in. the readme, and there is also a [macro](https://github.com/nholthaus/units#unit-definition-macros) that does some extra fanciness, but we don't need it here.

theVerySharpFlat, 2022-01-17:
Ok, so based on the documentation, this line of code:

https://github.com/rambunction4330/RobotCode2022/blob/7c8d7fa338a5ffff9e62b663a55127b85ac33822/lib/include/rmb/motorcontrol/talon/TalonSRXVelocityController.h#L21

means that for every 1 radian there are ``` 2π/4096 ``` of your unit. This really doesn't make sense to me. 

When I do the math, it is
```
2π rad * (2π rad / 4096 ticks) = 4 * π^2 rad / 4096 ticks        

instead of

2π rad * (4096 ticks / 2π rad) = 4096 ticks
```

So, I believe that the code should be defined like this:
``` cpp
using RawUnit = typename units::unit<std::ratio<4096, 2>, units::radians, std::ratio<-1>>;
```

As you would want to cancel out 2π to get 4096 ticks. [I read the docs!!!!](http://nholthaus.github.io/units/structunits_1_1unit.html)

gcjurgiel, 2022-01-17:
It's definitely possible, if not likely, that I defined the unit wrong. However, degrees is defined with `std::ratio<1, 180>` [here](https://github.com/nholthaus/units/blob/master/include/units.h#L3472) and the [inches example](https://github.com/nholthaus/units/blob/master/include/units.h#L847) used `std::ratio<1, 12>`. The ratio seems represent the conversion from the defined unit back to the base unit.

It is not that there are `2pi/4096` (or we should probably simplify it to `pi/2048`) ticks in every radian. It means that a single tick is `2pi/4096` radians. so `4096 ticks * (2pi rad/4096 ticks) = 2pi radians`.

closed 2022-01-17 by theVerySharpFlat