rambunction4330/RobotCode2022 #27: Initialization order warnings from compiler about MecanumDrive and HolonomicDrive code

theVerySharpFlat, 2022-01-20:
```
Run ./gradlew build
Downloading https://services.gradle.org/distributions/gradle-7.3.3-bin.zip
...........10%...........20%...........30%...........40%...........50%...........60%...........70%...........80%...........90%...........100%

Welcome to Gradle 7.3.3!

Here are the highlights of this release:
 - Easily declare new test suites in Java projects
 - Support for Java 17
 - Support for Scala 3

For more details see https://docs.gradle.org/7.3.3/release-notes.html

Starting a Gradle Daemon (subsequent builds will be faster)

> Task :compileFrcUserProgramReleaseExecutableFrcUserProgramCpp
Warnings in file VisionSubsystem.cpp ....
/__w/RobotCode2022/RobotCode2022/src/main/cpp/vision/VisionSubsystem.cpp: In member function 'Vector3<float> VisionSubsystem::GetHubPosition()':
/__w/RobotCode2022/RobotCode2022/src/main/cpp/vision/VisionSubsystem.cpp:45:14: warning: variable 'hubHeight' set but not used [-Wunused-but-set-variable]
         auto hubHeight = table->GetEntry("HeightToHub");
              ^~~~~~~~~
Warnings in file MecanumDrive.cpp ....
In file included from /__w/RobotCode2022/RobotCode2022/lib/src/drive/MecanumDrive.cpp:1:0:
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/MecanumDrive.h: In constructor 'rmb::MecanumDrive::MecanumDrive(rmb::VelocityController<units::unit<std::ratio<1>, units::base_unit<std::ratio<1> > > >&, rmb::VelocityController<units::unit<std::ratio<1>, units::base_unit<std::ratio<1> > > >&, rmb::VelocityController<units::unit<std::ratio<1>, units::base_unit<std::ratio<1> > > >&, rmb::VelocityController<units::unit<std::ratio<1>, units::base_unit<std::ratio<1> > > >&, frc::MecanumDriveKinematics, units::velocity::meters_per_second_t, units::angular_velocity::radians_per_second_t, frc::Gyro&)':
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/MecanumDrive.h:50:35: warning: 'rmb::MecanumDrive::maxRotVelocity' will be initialized after [-Wreorder]
       units::radians_per_second_t maxRotVelocity;
                                   ^~~~~~~~~~~~~~
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/MecanumDrive.h:47:18: warning:   'frc::Gyro& rmb::MecanumDrive::gyro' [-Wreorder]
       frc::Gyro& gyro;
                  ^~~~
/__w/RobotCode2022/RobotCode2022/lib/src/drive/MecanumDrive.cpp:4:3: warning:   when initialized here [-Wreorder]
   MecanumDrive::MecanumDrive(VelocityController<units::meters>& fl,
   ^~~~~~~~~~~~
Warnings in file HolonomicTrajectoryCommand.cpp ....
In file included from /__w/RobotCode2022/RobotCode2022/lib/src/drive/HolonomicTrajectoryCommand.cpp:3:0:
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/HolonomicTrajectoryCommand.h: In constructor 'rmb::HolonomicTrajectoryCommand::HolonomicTrajectoryCommand(frc::Trajectory, rmb::HolonomicDrive&, frc::HolonomicDriveController)':
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/HolonomicTrajectoryCommand.h:23:25: warning: 'rmb::HolonomicTrajectoryCommand::drive' will be initialized after [-Wreorder]
         HolonomicDrive& drive;
                         ^~~~~
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/HolonomicTrajectoryCommand.h:22:39: warning:   'frc::HolonomicDriveController rmb::HolonomicTrajectoryCommand::driveController' [-Wreorder]
         frc::HolonomicDriveController driveController;
                                       ^~~~~~~~~~~~~~~
/__w/RobotCode2022/RobotCode2022/lib/src/drive/HolonomicTrajectoryCommand.cpp:6:5: warning:   when initialized here [-Wreorder]
     HolonomicTrajectoryCommand::HolonomicTrajectoryCommand(frc::Trajectory t,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~

> Task :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
> Task :linkFrcUserProgramReleaseExecutable
> Task :frcUserProgramReleaseExecutable
> Task :checkFrcUserProgramDebugExecutable UP-TO-DATE
> Task :checkFrcUserProgramReleaseExecutable UP-TO-DATE
> Task :check UP-TO-DATE

> Task :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
Warnings in file VisionSubsystem.cpp ....
/__w/RobotCode2022/RobotCode2022/src/main/cpp/vision/VisionSubsystem.cpp: In member function 'Vector3<float> VisionSubsystem::GetHubPosition()':
/__w/RobotCode2022/RobotCode2022/src/main/cpp/vision/VisionSubsystem.cpp:45:14: warning: variable 'hubHeight' set but not used [-Wunused-but-set-variable]
         auto hubHeight = table->GetEntry("HeightToHub");
              ^~~~~~~~~
Warnings in file MecanumDrive.cpp ....
In file included from /__w/RobotCode2022/RobotCode2022/lib/src/drive/MecanumDrive.cpp:1:0:
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/MecanumDrive.h: In constructor 'rmb::MecanumDrive::MecanumDrive(rmb::VelocityController<units::unit<std::ratio<1>, units::base_unit<std::ratio<1> > > >&, rmb::VelocityController<units::unit<std::ratio<1>, units::base_unit<std::ratio<1> > > >&, rmb::VelocityController<units::unit<std::ratio<1>, units::base_unit<std::ratio<1> > > >&, rmb::VelocityController<units::unit<std::ratio<1>, units::base_unit<std::ratio<1> > > >&, frc::MecanumDriveKinematics, units::velocity::meters_per_second_t, units::angular_velocity::radians_per_second_t, frc::Gyro&)':
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/MecanumDrive.h:50:35: warning: 'rmb::MecanumDrive::maxRotVelocity' will be initialized after [-Wreorder]
       units::radians_per_second_t maxRotVelocity;
                                   ^~~~~~~~~~~~~~
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/MecanumDrive.h:47:18: warning:   'frc::Gyro& rmb::MecanumDrive::gyro' [-Wreorder]
       frc::Gyro& gyro;
                  ^~~~
/__w/RobotCode2022/RobotCode2022/lib/src/drive/MecanumDrive.cpp:4:3: warning:   when initialized here [-Wreorder]
   MecanumDrive::MecanumDrive(VelocityController<units::meters>& fl,
   ^~~~~~~~~~~~
Warnings in file HolonomicTrajectoryCommand.cpp ....
In file included from /__w/RobotCode2022/RobotCode2022/lib/src/drive/HolonomicTrajectoryCommand.cpp:3:0:
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/HolonomicTrajectoryCommand.h: In constructor 'rmb::HolonomicTrajectoryCommand::HolonomicTrajectoryCommand(frc::Trajectory, rmb::HolonomicDrive&, frc::HolonomicDriveController)':
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/HolonomicTrajectoryCommand.h:23:25: warning: 'rmb::HolonomicTrajectoryCommand::drive' will be initialized after [-Wreorder]
         HolonomicDrive& drive;
                         ^~~~~
/__w/RobotCode2022/RobotCode2022/lib/include/rmb/drive/HolonomicTrajectoryCommand.h:22:39: warning:   'frc::HolonomicDriveController rmb::HolonomicTrajectoryCommand::driveController' [-Wreorder]
         frc::HolonomicDriveController driveController;
                                       ^~~~~~~~~~~~~~~
/__w/RobotCode2022/RobotCode2022/lib/src/drive/HolonomicTrajectoryCommand.cpp:6:5: warning:   when initialized here [-Wreorder]
     HolonomicTrajectoryCommand::HolonomicTrajectoryCommand(frc::Trajectory t,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~

> Task :linkFrcUserProgramDebugExecutable
> Task :frcUserProgramDebugExecutable
> Task :assemble
> Task :build

BUILD SUCCESSFUL in 1m 15s
4 actionable tasks: 4 executed
```

gcjurgiel, 2022-01-20:
You're probably right. There are also some other issues in there as to what is passed into the constructor as a reference and what is passed as a value.

ADawesomeguy, 2022-01-20:
Working on a fix already :)

ADawesomeguy, 2022-01-20:
Fixed with #29.

closed 2022-01-20 by ADawesomeguy