rambunction4330/RobotCode2022 #32: Doxygen Documentation

theVerySharpFlat, 2022-01-21:
## Intro
As our repository gets larger (for some reason when I say that in my head it sounds like "as the pacific garbage patch gets larger...". Coincidence? ), we need to document our code so that we know what other people are doing and how to use their code (and the code that we write ourselves). Thus, we need to implement a documentation system - in our case, Doxygen. Now, since we are integrating a new tool into our workflow, we need to screw with ~~an improvised explosive device~~ the build system.

## Resources
* [The massive Doxygen manual](https://www.doxygen.nl/manual/index.html)
* There is a [Doxygen Gradle Plugin](https://plugins.gradle.org/plugin/org.ysb33r.doxygen) that google is recommending me. Its main-page is [here](https://gitlab.com/ysb33rOrg/doxygen-gradle-plugin)
* The wpilib project [is an example](https://github.com/wpilibsuite/allwpilib/blob/main/docs/build.gradle) of how to use this plugin

## Problems
1. We all have to learn Doxygen and learn how to document our code (ewwwww)
2. We need to mess with the build system (gross)
3. We need to mess with a foreign piece of software (disgusting)

ADawesomeguy, 2022-01-21:
What an abhorrent issue.

theVerySharpFlat, 2022-01-21:
@ADawesomeguy welcome to the party. It will be fun.

PR #46 (2022-01-22): Implement basic Doxygen setup
Tasks done:
- [x] Add Doxyfile
- [x] Add Doxygen GitHub pages workflow
- [x] Add initial example with HolonomicDriveOdometry.cpp
- [x] Add doxygen-awesome.css from jothepro's repository
- [x] Add basic credits to README

Resolves #32
closed 2022-01-22 by ADawesomeguy