rambunction4330/RobotCode2022 #33: SparkMax error checking

theVerySharpFlat, 2022-01-21:
Most, if not all, SparkMax ```SetSomething()``` functions return a `rev::REVLibError` (defined [here](https://codedocs.revrobotics.com/cpp/namespacerev.html#a3d1d3b447782f72a44fdf06eefad119d)), which I do not check for in the code currently. @gcjurgiel, do you think it would be benificial/worth-the-time to check for these errors and then print them out?

The code, in my mind looks like this:

```cpp
#define CHECK_REVLIB_ERROR(x) checkError(x, #x, __FILE__, __LINE__, __func__);

static bool checkError(rev::REVLibError error, const char* functionCall, const char* fileName, int line, const char* parentFunction) {
  switch(error) {
    case rev::RevLibError::kOk:
      // success
      return true;
    case rev::RevLibError::kError:
      // use the wpilib equivelent of std::cout. Need int -> string conversions
      std::cout << "in " << fileName << ":" << line << std::endl;
      std::cout << "in function \"" << parentFunction << "\"" << std::endl;
      std::cout << "\"" << functionCall << "\" generated a " << "REVLibError of type rev::RevLibError::kError" << std::endl;
      break;
    //....... all the other 9k error types
  }
  return false;
}
```

gcjurgiel, 2022-01-21:
That seems useful for debugging if you're willing to implement it. The WPILib stream you'd likely want use is `errs()` in `wpi/raw_ostream`

theVerySharpFlat, 2022-01-21:
Sounds good. Its going to be long and boring to implement, but it will certainly make me feel better about code that we can compile but not run. I also wish that the wpilib stdout functions could print things other std::strings or whatever string format they take.

ADawesomeguy, 2022-01-21:
@theVerySharpFlat I'd be happy to work with you on this while I'm working on the subsystem or once I finish.

theVerySharpFlat, 2022-01-21:
> @theVerySharpFlat I'd be happy to work with you on this while I'm working on the subsystem or once I finish.

Its really pretty easy, its just long and boring. All I need to do is check for each type of error in a switch statement an log the correct one. I also need to wrap all of the api calls in the macro. Since I did the rest of the SparkMax code, I think I should handle this one. However, the Talons still need to be wrapped with a position controller class, although I don't know if we are going to be using the Talons.

ADawesomeguy, 2022-01-21:
Alright no problem, just thought i could help speed it up a bit.

theVerySharpFlat, 2022-01-21:
Implemented. Will close when pull request is approved.

closed 2022-01-21 by gcjurgiel