rambunction4330/RobotCode2022 #35: Feedforward subclasses

gcjurgiel, 2022-01-21:
For anyone with nothing to do, implement subclasses of the Feedforward interface for all the WPILib feedforward types (frc::SimpleMotorFeedforward, frc::ElevatorFeedForward, and frc::ArmFeedforward), and implement them in the Velocity and Position Controllers.

theVerySharpFlat, 2022-01-21:
Best Feature Request EVER!!!!!!

theVerySharpFlat, 2022-01-26:
Technically this is done with the modifications I made today, but it remains untested.

gcjurgiel, 2022-01-26:
I created a branch with a quick implementation of a feedforward using references. I'm not quite sure how to link it, but I'm sure you can find it. I had to create a constant global variable (spooky) to reference of default arguments, but aside from that it's pretty much identical to what we had before. I fit looks good one of us can build it back up to what you made before.

gcjurgiel, 2022-01-26:
[here's a link](https://github.com/rambunction4330/RobotCode2022/compare/feedforward-George)

closed 2022-01-28 by gcjurgiel