rambunction4330/RobotCode2022 #48: Doxygen Workflow on PRs

ADawesomeguy, 2022-01-22:
It seems the workflow builds the page even for PRs to main, so I think we have a couple options:
1. We can allow these because it keeps the #47 PR constantly building
2. We can get rid of it and test the build at the very end

ADawesomeguy, 2022-01-23:
Seems to be working as it's only building pushes to `doxygen`. Closing.

theVerySharpFlat, 2022-01-24:
I feel like it should be building the documentation from the main branch not the doxygen branch as "main" is the official branch with the official code.

ADawesomeguy, 2022-01-24:
Well right now all the commits are going to the doxygen branch, so I was thinking it should follow that now and then once we merge the PR it'll follow the main branch.

theVerySharpFlat, 2022-01-24:
That makes sense. Thanks for the clarification!

closed 2022-01-24 by theVerySharpFlat