rambunction4330/RobotCode2022 #68: Command Based Overview

gcjurgiel, 2022-01-30:
We're going to use WPILibs command based programing for robot specific code. It should allow for more people to do more at once (thus preventing boredom). I highly recommend looking over WPILibs docs on command based programming [here](https://docs.wpilib.org/en/stable/docs/software/commandbased/index.html), but I'll highlight the main points.

Each component of the robot is encapsulated into a subsystem (drive subsystem, intake subsystem, shooter subsystem, etc). Each subsystem also have a `Periodic()` function that will be called repeated whenever the robot is on.

Actions the robot can perform are encapsulated in commands. Commands get scheduled through the `CommandScheduler`, which will manage their execution. Each command has three phases: `Initialization()`,` Execute()`, and `End()`. `Initialization()` ruins once when the command starts, `Execute` is called repeatedly through the life of the command, and `End()` is called once when the command has finished. Additionally, you must define a `IsFinished()` function that returns true once the command has completed. Commands will also take subsystems as requirements so as not to tell a component to do two things at once. If you are just getting information from a subsystem it does not need to be added to the requirements, but it should be marked as `const` (This last part is not in the WPILib guide, but I think it is a good model to follow). Commands can then triggered by joystick button presses, or linked together in `SequentialCommands` and `ParallelCommands`. This means complex commands can be broken down into their individual pieces and then linked together for simplicity.

These `Commands` and `Subsystems` are held in the `RobotContainer` to prevent the `Robot.h` file from getting too clogged up. Additionally button bindings are defined in this class.

The `Constants.h` file holds all the constants for the robot (motor ids, PID values, max/min values, etc). This can be subject to change. I think it is useful to hold the constants in a separate file since there are so many of them, but we may want to make a speared `Constants.h` file for each subsystem.

Each of us should probably take our own subsystem and corresponding commands for simplicity. It will hopefully give everyone something to do and prevent boredom. I might have Abheek formalize this in GitHub projects if ti would be useful (because I certainly don't know how to). I'll go over in more detail what I'm thinking for each subsystem with the person doing it next meeting, but **YOU DO NOT HAVE TO FOLLOW WHAT I SAY**. Its *your* code. Write it how *you* want to, and make it do what *you* think is useful. That being said, I am the dictator of this codebase and if your style doesn't align with my taste there will be severe consequences.

ADawesomeguy, 2022-01-30:
We be coding fr fr.

theVerySharpFlat, 2022-01-31:
"*YOU DO NOT HAVE TO FOLLOW WHAT I SAY*" - Jurj Jurjl 2022

gcjurgiel, 2022-01-31:
Mostly I'm just annoyed at everyone asking me what to do every five minutes. I clearly don't know what I'm doing or have a plan. You'll be better off not to listen to me. 

(To be clear, the first part is sarcasm. The rest... not so much).

notSam25, 2022-01-31:
Thanks for the feedback george, ill keep that in mind whenever I ask you what to do next, because you are the only one who has the slightest idea how to do this. 🦖 
[image]