rambunction4330/RobotCode2022 #80: The issue with the position controller

theVerySharpFlat, 2022-02-02:
@gcjurgiel !!!!!!!!!!! I think the issue with the position controller isn't an issue with the actual postion controller code! I think its because we didn't account for the gear ratio from the motor to the wheel. Thus, when we told it to go to position 2.x it was going to position 2.x, just on the motor and not the wheel, which is why the wheel didn't spin all the way around/

ADawesomeguy, 2022-02-02:
deep.

gcjurgiel, 2022-02-02:
I think your right! It's so annoying that it's always such a small error causing all our pain.

theVerySharpFlat, 2022-02-02:
and now we can't even fix it. All of the code is sitting on the laptop at school

ADawesomeguy, 2022-02-02:
You should always push smh. Btw would this be fixed with passing a gear ratio parameter or something?

gcjurgiel, 2022-02-02:
You'd fix it by making the conversion factor `3_in / 12_rad` instead of `3_in / 1_rad`.

gcjurgiel, 2022-02-02:
it's literally a one character difference

ADawesomeguy, 2022-02-02:
Vivimos en un sociedad.

theVerySharpFlat, 2022-02-02:
No, es una problema economica

theVerySharpFlat, 2022-02-02:
¡necesitamos mas dinero!

theVerySharpFlat, 2022-02-09:
We need to resolve merge issues on the linked pull request so we can close this issue.

gcjurgiel, 2022-02-09:
I already did, merged, and deleted the branch.

closed 2022-02-09 by gcjurgiel

ADawesomeguy, 2022-02-09:
Bruh Aiden be slackin. I bet he forgot to `git pull` 🤓

theVerySharpFlat, 2022-02-09:
We need to update `motto.md` for my stupidity