rambunction4330/RobotCode2022 #82: TODOS: please list

theVerySharpFlat, 2022-02-03:
Looking for things to do while we are confined to remote development.

gcjurgiel, 2022-02-04:
Here are a few things that you might be able to do:

- Setting up subsystems and commands (Specifically storage, shooter, and climber)
- Setting up the RobotContainer class
- Accelerometer odometry using the navX?
- Implementing ballistic trajectories in code (I've already done the math)
- Helping me with vision.
- Unfinished talon motor controller classes
- Documenting

Let me know what sounds most interesting so I can give you more specific details, or anything else you van think of in case I'm missing something.

theVerySharpFlat, 2022-02-04:
Vision sounds cool, but I think we have more important things to work on. I might look into the accelerometer odometry thing using the navX. It's going to be a lot of math, but I think it will be worth it.

theVerySharpFlat, 2022-02-04:
How soon are we going to get a robot with the subsystems that you mentioned? Don't we only have two weeks to test that code? Scary

gcjurgiel, 2022-02-04:
>It's going to be a lot of math

look at [this](https://www.kauailabs.com/public_files/navx-mxp/apidocs/c++/class_a_h_r_s.html#aae2d8d97068bb168c41346532593f626) it might not be

gcjurgiel, 2022-02-04:
The two week limit is a little loose though. We will have up until after spring break to actually get the robot working. I know Jiya doesn't want to be meeting that whole time, and I can't blame her.  It's a long time, and there will be nothing for anyone besides programming to do. We will have seminars though, and I can probably get us more after school meetings if we need it. I'm not too worried... yet.

theVerySharpFlat, 2022-02-04:
Wow.

Our hands are being held. So are our feet. We're getting frickin carried.

theVerySharpFlat, 2022-02-04:
> I'm not too worried... yet.

- @gcjurgiel 2022

theVerySharpFlat, 2022-02-04:
thanks George

gcjurgiel, 2022-02-04:
Accelerometer odometry might not even be worth pursuing though, looking at [this](https://pdocs.kauailabs.com/navx-mxp/wp-content/uploads/2019/02/navx-mxp_robotics_navigation_sensor_user_guide.pdf) (pg. 7).

gcjurgiel, 2022-02-04:
Though [newer versions](https://pdocs.kauailabs.com/navx-mxp/wp-content/uploads/2021/02/navx2-mxp_robotics_navigation_sensor_user_guide-9.pdf) (same page) are more accurate, but I'm not sure which we have.

gcjurgiel, 2022-02-04:
But the mecum encoders aren't all that great either, so it's worth testing

theVerySharpFlat, 2022-02-04:
Wait, so they didn't use the navx for odometry?

theVerySharpFlat, 2022-02-04:
Also, the displacement functions are experimental and the docs say that they are innacurate.

gcjurgiel, 2022-02-04:
They are. It cites ±1 meter over the 15 second auto, which is quite bad. If it interests you, it would be nice to experiment with it, but it likely won't be better than the encoder odometry so it's not a high priority. (Another place we could get odometry from is vision, but that is a hard problem. We likely don't have time for this season unless someone wants to devote a *lot* of time and learning to it)

gcjurgiel, 2022-02-04:
Odometry is useful for autonomously driving a single trajectory, not long term position tracking. Trajectories works similarly to a  profiledPID controller (for position). It needs a feedback position and velocity and odometry acts as that feedback. Following trajectories is not necessary for auto, but it can allow us to do more complex things during auto. 

But, after those 15 seconds odometry is mostly useless since human drivers can just see here the robot is. In past years it might have been beneficial to run trajectories during the teleop phase for better alignment, but it doesn't seem that way this year. We already have a good enough form of odometry implemented using encoders, so a new odometry system is not needed (although I do find it cool).

We will not need to autonomously find the hanger.

closed 2022-02-11 by gcjurgiel