rambunction4330/RobotCode2022 #101: Mechanism Characterization

gcjurgiel, 2022-02-17:
We need to calculate the feedforward and PID constants for all the mechanisms so that we can use the fancy code we have been working on. After we have the constants we can start writing/testing commands and subsystems.

- [ ] Drivetrain
- [ ] Intake (spinner optional)
- [ ] Turret
- [ ] Hood
- [ ] Flywheel

ADawesomeguy, 2022-02-17:
Why isn't Aiden assigned this.

gcjurgiel, 2022-02-17:
I forgot to

theVerySharpFlat, 2022-02-21:
[04b8170](https://github.com/rambunction4330/RobotCode2022/pull/98/commits/04b8170c8acbbda24428b00e81f60b6a49ee05d0) updated the constants. @gcjurgiel could you make sure I did it correctly

theVerySharpFlat, 2022-02-23:
mechanical added weight to the turret. need to recalculate the PID and FF values

notSam25, 2022-02-25:
> I forgot to

you forgor 💀

theVerySharpFlat, 2022-03-04:
I broke the flywheel trying to get feedforward values :(

theVerySharpFlat, 2022-03-06:
We did the hood

theVerySharpFlat, 2022-03-07:
@gcjurgiel we did the drivetrain today right?

gcjurgiel, 2022-03-08:
Yes!

closed 2022-03-08 by gcjurgiel