rambunction4330/RobotCode2022 #108: FIX YOUR COMMIT MESSAGES

ADawesomeguy, 2022-02-22:
They're not making sense.

ADawesomeguy, 2022-02-22:
`fix` - Aiden
`merge` - Sam

yall disappoint me

theVerySharpFlat, 2022-02-22:
@ADawesomeguy could you please provide us with more info? Where are these commits of mine?

ADawesomeguy, 2022-02-22:
Literally the first commit in https://github.com/rambunction4330/RobotCode2022/pull/98

theVerySharpFlat, 2022-02-23:
I think I may have been spiting Josh Rosen

ADawesomeguy, 2022-02-23:
Bruv you gotta step up your game.

theVerySharpFlat, 2022-02-23:
In spiting people or writing commit messages?

ADawesomeguy, 2022-02-23:
Both

notSam25, 2022-02-25:
idc about your stupid git names. I named my commits after what Aiden told me to. Sugma

theVerySharpFlat, 2022-02-25:
> idc about your stupid git names. I named my commits after what Aiden told me to. Sugma

Why blame on me? 

I think the thing these days is to shoot an insult back at the attacker.

Well

You know how we count your attendance? The amount of soda that we have in the fridge.

[image]

So take that.

ADawesomeguy, 2022-02-25:
Gotem

notSam25, 2022-02-25:
Thats it, im not giving anyone anymore shmuchks offbrand soda. because of Aiden's smol brain nobody gets anymore.

notSam25, 2022-02-25:
rip bozo

theVerySharpFlat, 2022-02-25:
> rip bozo

well you are stupid head too :(

theVerySharpFlat, 2022-02-25:
gay pride

theVerySharpFlat, 2022-02-25:
I hate them gayz

closed 2022-02-25 by theVerySharpFlat