rambunction4330/RobotCode2022 #116: Compilation database generation.

theVerySharpFlat, 2022-02-27:
## TL; DR: Aiden wants to use a gradleRIO hack so that he doesn't have to use vscode

When you build, gradle will generate `options.txt` files that it dumps in the `build/tmp/compileFrcUserProgramDebugExecutableFrcUserProgramCpp` and `build/tmp/compileFrcUserProgramReleaseExecutableFrcUserProgramCpp` folders. 

The contents of these files looks something like this:
<details>
<summary> expand </summary>

```
-x
c++
-c
-Wall
-Wextra
-std=c++17
-Wformat=2
-pedantic
-Wno-psabi
-Wno-unused-parameter
-Wno-error=deprecated-declarations
-fPIC
-rdynamic
-pthread
-Og
-g
-nostdinc
-I
/workspace/RobotCode2022/src/main/include
-I
/workspace/RobotCode2022/lib/include
-I
/workspace/.gradle/caches/transforms-3/41aaea03c19142e83441d9eb14367ea3/transformed/wpilibNewCommands-cpp-2022.2.1-headers
-I
/workspace/.gradle/caches/transforms-3/12c16b9377b895ccfc45f3f12ee1041c/transformed/REVLib-cpp-2022.1.1-headers
-I
/workspace/.gradle/caches/transforms-3/a55d9b918e6caa2d33f80bea99273f38/transformed/REVLib-driver-2022.1.1-headers
-I
/workspace/.gradle/caches/transforms-3/2b9ce387dc001246545daba3150c0417/transformed/wpiapi-cpp-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/445ba3bbfd7ddba68a0dbf6943a7132f/transformed/api-cpp-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/cb4de00697ee2fe0d4c781bafa620eaf/transformed/cci-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/80635fd6694bbc8400abe0538282f8d0/transformed/wpiapi-cpp-sim-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/173fb1102aafca56090092e10923c6e6/transformed/api-cpp-sim-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/95fe0d2ca9afbe758348eb112f367d17/transformed/cci-sim-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/8855ad2be18847ad7071775259d1e129/transformed/simTalonSRX-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/d5b1b53d6388c7b5639e7227d88495d2/transformed/simTalonFX-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/faa761211ef18beca2190bca42ddb8ab/transformed/simVictorSPX-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/b27847e97d5b93129d10c78a0b5b1a27/transformed/simPigeonIMU-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/802d5bcfd0415b4dc8b1254a88153224/transformed/simCANCoder-5.20.2-headers
-I
/workspace/.gradle/caches/transforms-3/0546f9ba5633d276349366971ea85de2/transformed/navx-cpp-4.0.442-headers
-I
/workspace/.gradle/caches/transforms-3/2c33e55f4a00175bd8c17633c6f39c89/transformed/wpilibc-cpp-2022.2.1-headers
-I
/workspace/.gradle/caches/transforms-3/6d23c1fe2004456cf706e3c41cd803fd/transformed/ntcore-cpp-2022.2.1-headers
-I
/workspace/.gradle/caches/transforms-3/516bfd2a0a9cace0d8be91fba6d6618c/transformed/hal-cpp-2022.2.1-headers
-I
/workspace/.gradle/caches/transforms-3/421c2a3cc99623b3aca6609d65980626/transformed/wpimath-cpp-2022.2.1-headers
-I
/workspace/.gradle/caches/transforms-3/828d385f88f8274d023de1086f1dd58b/transformed/wpiutil-cpp-2022.2.1-headers
-I
/workspace/.gradle/caches/transforms-3/82660f90536b60cc6dcc3333182a5d33/transformed/netcomm-2022.2.3-headers
-I
/workspace/.gradle/caches/transforms-3/e0472765e657721c12c281be95841751/transformed/chipobject-2022.2.3-headers
-I
/workspace/.gradle/caches/transforms-3/3d5dace4cb6ae2a9a26cd9ba78c9fe55/transformed/visa-2022.2.3-headers
-I
/workspace/.gradle/caches/transforms-3/298ce67bdf85f6f56a9b648cedf7e102/transformed/cameraserver-cpp-2022.2.1-headers
-I
/workspace/.gradle/caches/transforms-3/caf4cb833c912ca0d9d59cb41941f5b9/transformed/cscore-cpp-2022.2.1-headers
-I
/workspace/.gradle/caches/transforms-3/b23233c12c84ef29bfe67b931ca5cd8d/transformed/opencv-cpp-4.5.2-1-headers
-isystem
/home/gitpod/.gradle/toolchains/frc/2022/roborio/arm-frc2022-linux-gnueabi/usr/lib/gcc/arm-frc2022-linux-gnueabi/7.3.0/include
-isystem
/home/gitpod/.gradle/toolchains/frc/2022/roborio/arm-frc2022-linux-gnueabi/usr/lib/gcc/arm-frc2022-linux-gnueabi/7.3.0/include-fixed
-isystem
/home/gitpod/.gradle/toolchains/frc/2022/roborio/arm-frc2022-linux-gnueabi/usr/include/c++/7.3.0
-isystem
/home/gitpod/.gradle/toolchains/frc/2022/roborio/arm-frc2022-linux-gnueabi/usr/include/c++/7.3.0/arm-frc2022-linux-gnueabi
-isystem
/home/gitpod/.gradle/toolchains/frc/2022/roborio/arm-frc2022-linux-gnueabi/usr/include/c++/7.3.0/backward
-isystem
/home/gitpod/.gradle/toolchains/frc/2022/roborio/arm-frc2022-linux-gnueabi/usr/include
```

</details>
Obviously it's just a big list of compiler options, which means we can generate a compile_commands.json file using this data, which means that we can use editors/ides like vim or clion. Additionally, this means we can probably get better c++ completion in vscode itself because I've noticed that the frc completion is pretty subpar.

The compilation database documentation is pretty simple. [Here](https://clang.llvm.org/docs/JSONCompilationDatabase.html) it is. 

This is something I would really like to work on in my free time.

theVerySharpFlat, 2022-02-27:
Need to find where wpi keeps its compiler though

theVerySharpFlat, 2022-02-27:
It seems to be (not on windows) at `$HOME/.gradle/toolchains/frc/2022/roborio/bin/arm-frc2022-linux-gnueabi-g++` assuming it was installed through `./gradlew installRoboRioToolchain`. I think it might be installed in a different location for the "proper" install.

gcjurgiel, 2022-02-27:
I'm not entirely clear on what specifically you want to do, but it's definitely something to look into. Gradel and VScode can kinda suck, and more options would be nice. I know that Noah used to be able to deploy from InteliJ, but I think he was still using the same commands as in VScode. I also know that frc supports some level of Cmake compilation, but that might only be for installing the tools. As of for the compiler, I think I found it in `~/wpilib/[YYYY]/roborio/bin/`

That being said, it should be a goal for the offseason or next year. It's too far into the season to completely overhaul our build system.

theVerySharpFlat, 2022-02-27:
> That being said, it should be a goal for the offseason or next year. It's too far into the season to completely overhaul our build system.

This wouldn't be overhauling the build system. 

What I am doing is trying to generate a compilation database based on the output of gradle. The plan isn't to make a change to gradle. It's to get better intellisense so I can use clion or vim.

Since gradle dumps all of its compile options into a text file after build/configure, I can parse that file to generate a compile_commands.json file which will list all of the source files and the commands required to compile them. IDEs and editors then use this to generate code completion and code insights. 

This is essentially going to be a script that I write that you would then run after compilation to get code completion in a different editor.

gcjurgiel, 2022-02-27:
Thanks for the explication. That sounds great!

notSam25, 2022-02-28:
I hate VSCODE, thanks for the alternative aids.

theVerySharpFlat, 2022-02-28:
[here's the completed project](https://github.com/theVerySharpFlat/gradleRIO_CompilationDatabase). Still need to add cliargs.

theVerySharpFlat, 2022-02-28:
> I hate VSCODE, thanks for the alternative aids.

doesn't work on windows. use the gradle plugin

theVerySharpFlat, 2022-02-28:
@gcjurgiel @ADawesomeguy check it out

ADawesomeguy, 2022-02-28:
sexy